hadabot/hadabot_main #5: Using mobile phone to control hadabot from teleop page

dsryzhov, 2020-10-06:
It would be great to have an option to control hadabot from mobile phone using teleop page. 

Teleop page is connected to ros2 on localhost.
It prevents from using teleop page to control hadabot from mobile phone (connected to the same LAN as ros2 computer)
In case teleop page would provide configuration option for ros2 IP address in the LAN
it could be possible to use teleop from mobile phone. 

Thanks for your work!

jackpien, 2020-10-16:
Hi @dsryzhov - interesting use case.

Just to clarify - you are suggesting a text field in the teleop HTML page from www.hadabot.com that allows the user to specify a specific IP address (instead of hard coding to localhost) to the ros_web_bridge server. Is that what you are thinking?

dsryzhov, 2020-10-17:
yes, it was my thought.