hadabot/hadabot_main #7: Codeserver does not work at the last version

dsryzhov, 2020-10-06:
Today I've upgraded to the last version of hadabot repo.
After that codeserver stoped work. Page with Visual Studio does not loading on port 9123
I use command 
docker-compose up -d 
to start codeserver and webbridge containers.

WebBridge works well

dsryzhov, 2020-10-09:
have tried codeserver one more time after last post. does not work.
all three ports are listened but return empty response

wget http://localhost:9123 writes connected
HTTP request sent, awaiting response.... No data received 
Retrying.

jackpien, 2020-10-16:
Hi @dsryzhov 

In Dockerfile.ros2_codeserver, I rebuilt everything from the line "# New way to install code server" and it looks like everything is working fine.

I'm not sure how fast you access http://localhost:9123 from the moment you run "docker-compose up -d" but the container does take a minute or so to get up and running. If you access http://localhost:9123 too quickly, then you won't get a server response.

Here's the code-server version FYI.

code-server: v3.4.1
VS Code: v1.45.1
Commit: 48f7c2724827e526eeaa6c2c151c520f48a61259
Date: 2020-06-04T10:39:14Z (4 mos ago)
Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/85.0.4183.102 Safari/537.36

Some final questions to see if there's anything else I'm missing:

1. What host system are you building the Hadabot containers on?
1. Did you wait a min or 2 before accessing http://localhost:9123?

Unless you have additional insights, I'll keep this open for another week before I close it.

Thanks!
Jack

dsryzhov, 2020-10-17:
>    2. Did you wait a min or 2 before accessing http://localhost:9123?

I build your image on ubuntu 20 installed on wsl2 in windows 10.
I check all three ports that are exposed from container. They are listened.
but does not receive any html page. I waited quite a lot I think.

I will try your image one more time.

 I've built another docker image with vnc and ros2. It works fine

dsryzhov, 2020-10-22:
Started new build of docker images. Will try one more time after they are finished.

dsryzhov, 2020-10-23:
Retested. It works!
May be I had not rebuild docker image before or had no patience to wait when container were started.
It takes very long time to build image and run the container. 
Nevertheless it was my mistake creating this issue.

closed 2020-10-23 by dsryzhov

jackpien, 2020-10-23:
Glad it worked @dsryzhov - the images do take a long time to build. ROS is a beast. One reason why I'm trying to create Hadabot so it abstracts out a lot of the setup headaches from users. 

Thanks so much for your help. Please keep the issues coming.