hadabot/hadabot_main #9: Formulas to compute distance traveled for each wheel

dsryzhov, 2020-10-07:
I assume that formulas used in code to compute distance traveled for each wheel is incorrect
No need to divide by PI
    // Compute distance traveled for each wheel
    float d_left_m = (wheel_radps_left_ * dt_s * wheel_radius_m_) / PI;
    float d_right_m = (wheel_radps_right_ * dt_s * wheel_radius_m_) / PI;

Formulas in blog post is also incorrect
https://blog.hadabot.com/implement-ros2-odometry-using-vscode-in-web-browser.html
No need to multiply by 2
    float d_left_m = 2* (wheel_radps_left_ * dt_s * wheel_radius_m_);
    float d_right_m = 2 * (wheel_radps_right_ * dt_s * wheel_radius_m_);

The correct formulas should be 
    float d_left_m = (wheel_radps_left_ * dt_s * wheel_radius_m_);
    float d_right_m = (wheel_radps_right_ * dt_s * wheel_radius_m_);

Let's check it
if wheel_radps_left_ = 2 * pi (one rotation per second)
distance should be equal 2*pi*wheel_radius_m
Using last formulas we obtainb correct values

jackpien, 2020-10-22:
Hi @dsryzhov 

You presented two issues:

------

(1) The code in .../p6/.../hadabot_odom.cpp  may be incorrect, specifically 

```
// Compute distance traveled for each wheel
float d_left_m = (wheel_radps_left_ * dt_s * wheel_radius_m_) / PI;
```

This is definitely not right but correct equation is not what you suggested. The correct equation is:

```
float d_left_m = (wheel_radps_left_ * dt_s * wheel_radius_m_) / (2*PI);
```

If the wheel is rotating one rotation per second, then:

* wheel_radps_left = (2 * PI) per second = 2*3.14 per second = 6.28 radians per second
* dt_s = 1 second
* wheel_radius_m = 1 meter

Result is d_left_m = (6.28 * 1 * 1)/(2*3.14) = 1 meter

Maybe you had thought wheel_radps is in rotations per second. But it is not - wheel_radps is in "[radians per second](https://en.wikipedia.org/wiki/Radian)".

--------

(2) Equation in [Implement ROS2 odometry using VSCode running in a web browser blog article](https://blog.hadabot.com/implement-ros2-odometry-using-vscode-in-web-browser.html) is incorrect:

You suggest:
> No need to multiply by 2
> float d_left_m = 2* (wheel_radps_left_ * dt_s * wheel_radius_m_);
> float d_right_m = 2 * (wheel_radps_right_ * dt_s * wheel_radius_m_);

The blog equation definitely looks wrong.

But let's first agree on (1) above which will result in correct equation for (2)

-------

**QUESTION** - Do you agree with (1) that my suggestion for the fix for the equation in the .../p6/.../hadabot_odom.cpp code is correct?

```
float d_left_m = (wheel_radps_left_ * dt_s * wheel_radius_m_) / (2*PI);
```

Thanks Jack

closed 2020-10-22 by jackpien

dsryzhov, 2020-10-22:
Great!

commit c046eda9 (2020-10-22): Fixes #9 - when computing wheel distance, do not need to divide by PI