hadabot/hadabot_main #13: Hadabot Micro-ROS based hardware

dsryzhov, 2020-10-23:
Hi Jack,

I've published my Hadabot Micro-ROS based firmware.
https://github.com/dsryzhov/hadabot_rds

See description on the link how to configure and build it 
https://github.com/dsryzhov/hadabot_rds/tree/master/hadabot_microros_firmware

It took some time on this week 
- to refactor the original firmware version to use high-level arduino-esp32 API (originally low level esp idf api was used)
- to create docker image that helps to build the firmware easily

I assume that currently you can build and run it. 
On the next step I plan to additionally test and enhance 
the firmware after refactoring (to high level hardware api).

Regards,
Dmitry

dsryzhov, 2020-10-23:
Today Micro-ROS team has published precompiled library for Arduino IDE
https://github.com/micro-ROS/micro_ros_arduino
Esp32 architecture is not supported yet in micro-ros arduino library.
But I think it will be added in near time.

Also only serial transport is supported currently (no UDP).

jackpien, 2020-10-23:
This is fantastic @dsryzhov 

**Question** - do you know if the agent can be run inside a container in the Hadabot docker-compose stack? This way, "docker-compose up -d" will also launch the agent?

dsryzhov, 2020-10-23:
Hi Jack,

Please write me when you are succeded in running microros firmware on the
hadabot

Thanks for your advice. Yes, I know
I've created today docker-compose with all containers for Hadabot.

https://github.com/dsryzhov/hadabot_rds/blob/master/ros_ws/docker/docker-compose.yaml

Regards,
Dmitry

jackpien, 2020-10-30:
Hi Dmitry, this is fantastic work w micro-ROS (aka uROS). The learning curve to set up a build environment for uROS + ESP32 is not easy by any means. 

I'm in the process of following your instructions to build and play around with it. 

In the meantime - quick question to understand your design objectives: The main motivation for you to work on a uROS variation for the Hadabot firmware was the ROS message latency in using MicroPython as the firmware base? 

Were there any other reasons that made the MicroPython firmware functionally sub-optimal?

Thanks!
Jack

dsryzhov, 2020-10-30:
Hi Jack,

Yes it was the main reason.
I've started work with Micro-Ros when I've found that when Micropython is
used my car moves about 1 meter after a stop command issued.
I was also interested to enhance accuracy estimation with rotation sensors.

This week I've started to published timestamped messages with wheels
angular velocities to the ros.
As a result position estimation became much more accurate for linear motion.
I command to the car to move forward for 2 meters and return it to the same
point.
Difference between start and final position is about 1-5 cm.
When rotation is used there are problems with theta angle estimation. It's
not very accurate yet.
I plan to implement some improvements and tests what I can do.

Now I understand more about timings with Micro-Ros and think about moving
position estimation to the esp32.
and publish to the ros {x, y, theta, time}

Regards,
Dmitry Ryzhov

dsryzhov, 2020-10-30:
Using docker image it's not very hard.

I've solved some bug this week with Micro-Ros settings.
Following my previous instruction you will find that
esp32 trying to connect to 127.0.0.1 when started but not IP you have
configured.
I will update instructions.
By the way what OS do you use? I will help me understand your environment.

jackpien, 2020-10-30:
Hi @dsryzhov 

> Difference between start and final position is about 1-5 cm. When rotation is used there are problems with theta angle estimation. It's not very accurate yet. I plan to implement some improvements and tests what I can do.

I think you may be using the rising or falling edge to trigger ISR for wheel encoders. Is that right? If so, the signal out of the sensor is extremely noisy, even cleaned with the capacitor (ie low pass filter). The noisy signal can cause the ISR to trigger even though it is not a large rising edge. That is why I resorted to polling and reading the voltage directly in the original MicroPython firmware. I am experimenting with using a schmit trigger / feedback driven op-amp to make the signal totally square (near zero noise). 

> By the way what OS do you use? I will help me understand your environment.

I primarily develop on an Ubuntu OS machine. Also work on a MacOS but primarily on Ubuntu. 

**Question** - You use windows right? How did you get the Windows 10 system to find the ESP32? Did you have to install the CP120x driver? Some Hadabot users having trouble with Windows 10. How do you know which COM port Windows connects to when ESP32 plugged in? Sorry, I am a Windows rookie! =)

dsryzhov, 2020-10-31:
1. You are right. I had the same trouble this week. with
spurious interrupts.

But now I think that it's not a signal problem but ESP32 hardware + HAL
problem
On oscilloscope I see quite good signal from sensors.
When I've added capacitors (104) as in your Hadabot build the positive edge
became
not vertical a little bit. It think that not vertical positive edge also
causes additional interrutpts.

There are a lot of issues on the internet about such interrupts.
https://github.com/espressif/arduino-esp32/issues/1250
ESP triggers an interrupt event if signal level changes.
Moreover ESP-IDF library and Arduino library does not work correctly with
detecting
only positive edges or negative edges.
They work stable only when detecting both.

I think I've solved this problem for this case
by analyzing level change at the interrupt handler by myself.
At every interrupt I read signal level and compare it to the previous
stored valuе.
If level is changed I store new level and calculate duration and
process that signal change.

At the previous versions of firmware I processed both edges (positive
and negative) and calculated angular velocity on
every change. This solution gives more frequent measurements. It's good for
acceleration sections of the path and for rotation.
But this solution needs calibration of the sensor. Ratio of pulse durations
for high level and low level is needed
to calculate angular velocity for each signal change.

At the current version I use only positive edges to calculate angular
velocity. This approach does not need calibration.
But gives measurements 2 times frequently. It works well for linear motion
but for accelerated motion (and rotation)
I think it's not enough. I think I will return to measure angular velocity
at each change.

And I want to try integrate this measurements on the esp32 and publish
position to ros (not odometry message but simple 2D position).

I have a question for you. I saw that in Hadabot hardware rotation sensors
are installed with some rotation.
What is the reason for that ? There were no holes in standard parts ?

2. Ok, you are on Ubuntu

In this case you can flash and monitor easily from docker container with
esp idf sdk.
So you will be able to start container using idf.sh script and use
standard commands
idf.py build monitor flash in the container

I will add /dev:/dev mapping for docker run in idf.sh for monitor and flash
commands to work in the container.

On Windows 10 docker is installed on windows linux subsystem version 2
(WSL2).
We can install any number of lunux OS (from Windows Store) and use them in
command prompt in Windows.
When docker is installed it install linux in WSL2 and inside this linux run
docker containers.
I can use docker cli from windows but I've installed another WSL2 linux
distro - Ubuntu and use it for development.
I run windows terminal, write ubuntu command and that's it - I can use
ubuntu.
Ubuntu filesystem is mapped to net file disk on windows.

There is a problem with WSL2 Ubuntu now. It does not see Winfows com ports
at all.
Before I used to flash and monitor from Windows machine.
This week I've found the solution. I ran utility that exposes windows com
port to tcp port.
After that I am able to use idf.sh to start flash and monitor from docker
container via this tcp port

I don't remember problems with driver. I assume it was installed
automatically.
I can use windows device manager to see available COM ports or Ardiono IDE
where I see new port is added when esp32 plugged in.
In fact these port port is stable. Even if I change usb port that esp32 is
pluged in the com port number does not change.

Regards,
Dmitry

dsryzhov, 2020-10-31:
You can see my current implementation of interrupt here
https://github.com/dsryzhov/hadabot_rds/blob/master/hadabot_microros_firmware/components/rds/libraries/RotSensor/src/rotsensor.cpp

dsryzhov, 2020-10-31:
In the last version I've started to publish timestamped messages and use
timestamps in odometry update.
It makes odometry estimation much more accurate.
By the way I publish sensor measurements every 20 ms and update odometry on
20 ms also.

New messages
/hadabot/wheel_radps_right_timestamped
/hadabot/wheel_radps_left_timestamped

https://github.com/dsryzhov/hadabot_rds/blob/master/hadabot_microros_firmware/main/microros_interface.cpp

I'am not tried yet how to add new message types for Micro-ROS.
For experiment I use standard sensor_msgs__msg__Temperature
but not std_msgs__msg__Float32 to be asble to add timestamp.

https://github.com/dsryzhov/hadabot_rds/blob/master/hadabot_microros_firmware/main/microros_interface.cpp

In hadabot controller I subscribe to timestamped messages and use
timestamps in odometry_update to calculate positions
https://github.com/dsryzhov/hadabot_rds/blob/master/ros_ws/src/hadabot_driver/src/hadabot_controller.cpp

In the future I think about moving position calculation to esp32.
It will make it possible to update position more frequently (on every
angular velocity measurement)
At the same time it will make it possible to publish position to ros less
frequently.

dsryzhov, 2020-10-31:
I've updated the instruction how to use Micro-ROS based firmware
https://github.com/dsryzhov/hadabot_rds/tree/master/hadabot_microros_firmware

Changed it to using all idf.py commands inside docker container.
You can also use idf.sh script to run commands outside docker container as
before.