hadabot/hadabot_main #17: ROS2-web-bridge container fails after starting

agillies8, 2021-02-02:
To re-create:
1) Followed instructions to start demo of hadabot system by following these instructions:

$ git clone https://github.com/hadabot/hadabot_main.git
$ cd hadabot_main/docker
$ docker-compose up -d

2) Navigated to a new tab, opened localhost:9123, and can see the browser VScode terminal, and everyhing appears to be working, great

3) Open up a new tab at:  https://www.hadabot.com/tools/teleop.html, expecting to see a confirmation that it is connecting to the ROS2 codeserver container via the web-bridge container, however connection fails.

4) Upon further investigation of the log files, the web-server container has exited with this error message:

`/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/context.js:47
  static _instances = [];
                    ^

SyntaxError: Unexpected token =
    at new Script (vm.js:79:7)
    at createScript (vm.js:251:10)
    at Object.runInThisContext (vm.js:303:10)
    at Module._compile (internal/modules/cjs/loader.js:657:28)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)`

TO FIX:

To address this, I believe it is caused by the web-server container building nodejs version 10.x.x ( see this issue: https://github.com/RobotWebTools/ros2-web-bridge/issues/172)

I therefore changed line 11 in "Dockerfile.ros2_web_bridge" to:
    'curl -sL https://deb.nodesource.com/setup_12.x -o nodesource_setup.sh && \'

I then rebuilt the container, started it, and now I have a working connection!

Suggest taking a look at this and updating the repo to address this.

Cheers!

jackpien, 2021-02-02:
Hi @agillies8 - thanks or catching this! 

I have not verified the problem yet (since I tend to not rebuild my images often). 

But if you have the hadabot_main repo forked and are comfortable with pull requests, perhaps make your suggested change above and send me a pull request for that change. 

Thanks!
Jack

jackpien, 2021-02-05:
@agillies8 - never mind! I added the fix already.

If you find any other problems, please don't hesitate to file an issue.

Thanks!
Jack

commit 07222720 (2021-02-05): Fixes #17: for the ros2 web bridge, use node 12.x instead of 10.x as suggested by https://github.com/RobotWebTools/ros2-web-bridge/issues/172
closed 2021-02-05 by jackpien