hadabot/hadabot_main #25: Unsigned package errors in `new-user-software-stack-setup` tutorial

jschrier, 2022-08-23:
When on Step 4 of the [Setup the software stack tutorial](https://www.hadabot.com/new-user-software-stack-setup.html)

The `docker-compose up` step fails with the following errors:

```
#6 8.751 Reading package lists...
#6 9.952 W: GPG error: http://dl.google.com/linux/chrome/deb stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 4EB27DB2A3B88B8B
#6 9.952 E: The repository 'http://dl.google.com/linux/chrome/deb stable InRelease' is not signed.
------
executor failed running [/bin/sh -c apt-get update &&     apt-get upgrade -y &&     apt-get install -y locales curl gnupg2 lsb-release wget git sudo unzip &&     locale-gen en_US en_US.UTF-8 &&     update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 &&     export LANG=en_US.UTF-8 &&     curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -  &&     sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list' &&     rm -rf /var/lib/apt/lists/*]: exit code: 100
ERROR: Service 'hadabot_ros2_codeserver_service' failed to build : Build failed
```

jackpien, 2022-08-23:
Hi @jschrier - give the latest pull / commit a try.

closed 2022-08-23 by jackpien

jschrier, 2022-08-23:
Nope, @jackpien, using ce7505e still appears to be broken (I also tried both without and with a VPN, just to see if it would help)

```
#6 63.51 W: GPG error: http://dl.google.com/linux/chrome/deb stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 4EB27DB2A3B88B8B
#6 63.51 E: The repository 'http://dl.google.com/linux/chrome/deb stable InRelease' is not signed.
```