hadabot/hadabot_main #31: docker compose up: ImportError: cannot import name 'escape' from 'jinja2'

kingb, 2023-05-24:
Following the directions of running `docker compose up` in the `docker` subdirectory leads to this python error:

```
Attaching to hadabot_ros2_codeserver, hadabot_ros2_web_bridge
hadabot_ros2_codeserver  | * enable custom user: hadabot
hadabot_ros2_codeserver  | useradd: user 'hadabot' already exists
hadabot_ros2_codeserver  |   set default password to "ubuntu"
hadabot_ros2_codeserver  | cp: cannot stat '/root/.config': No such file or directory
hadabot_ros2_web_bridge  | Websocket started on ws://localhost:9090
hadabot_ros2_codeserver  | 2023-05-24 19:19:20,367 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
hadabot_ros2_codeserver  | 2023-05-24 19:19:20,367 INFO Included extra file "/etc/supervisor/conf.d/codeserver.conf" during parsing
hadabot_ros2_codeserver  | 2023-05-24 19:19:20,367 INFO Included extra file "/etc/supervisor/conf.d/supervisord.conf" during parsing
hadabot_ros2_codeserver  | 2023-05-24 19:19:20,371 INFO RPC interface 'supervisor' initialized
hadabot_ros2_codeserver  | 2023-05-24 19:19:20,371 CRIT Server 'unix_http_server' running without any HTTP authentication checking
hadabot_ros2_codeserver  | 2023-05-24 19:19:20,372 INFO supervisord started with pid 33
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,376 INFO spawned: 'nginx' with pid 35
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,382 INFO spawned: 'web' with pid 36
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,388 INFO spawned: 'codeserver' with pid 37
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,397 INFO spawned: 'xvfb' with pid 38
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,403 INFO spawned: 'wm' with pid 39
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,408 INFO spawned: 'lxpanel' with pid 40
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,412 INFO spawned: 'pcmanfm' with pid 41
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,415 INFO spawned: 'x11vnc' with pid 45
hadabot_ros2_codeserver  | 2023-05-24 19:19:21,419 INFO spawned: 'novnc' with pid 47
hadabot_ros2_codeserver  | Traceback (most recent call last):
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 122, in <module>
hadabot_ros2_codeserver  |     main()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 118, in main
hadabot_ros2_codeserver  |     entrypoint()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 78, in run_server
hadabot_ros2_codeserver  |     from vnc.app import app
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/vnc/app.py", line 6, in <module>
hadabot_ros2_codeserver  |     from flask import (
hadabot_ros2_codeserver  |   File "/usr/local/lib/python3.8/dist-packages/flask/__init__.py", line 14, in <module>
hadabot_ros2_codeserver  |     from jinja2 import escape
hadabot_ros2_codeserver  | ImportError: cannot import name 'escape' from 'jinja2' (/home/hadabot/.local/lib/python3.8/site-packages/jinja2/__init__.py)
```

DaveCollinsJr, 2023-08-09:
I am having the same issue.  Is anyone monitoring the Hadabot emails or this repo?

`hadabot_ros2_codeserver  | 2023-08-09 02:06:50,408 INFO spawned: 'web' with pid 208
hadabot_ros2_codeserver  | Traceback (most recent call last):
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 122, in <module>
hadabot_ros2_codeserver  |     main()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 118, in main
hadabot_ros2_codeserver  |     entrypoint()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 78, in run_server
hadabot_ros2_codeserver  |     from vnc.app import app
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/vnc/app.py", line 6, in <module>
hadabot_ros2_codeserver  |     from flask import (
hadabot_ros2_codeserver  |   File "/usr/local/lib/python3.8/dist-packages/flask/__init__.py", line 14, in <module>
hadabot_ros2_codeserver  |     from jinja2 import escape
hadabot_ros2_codeserver  | ImportError: cannot import name 'escape' from 'jinja2' (/home/hadabot/.local/lib/python3.8/site-packages/jinja2/__init__.py)
hadabot_ros2_codeserver  | 2023-08-09 02:06:51,081 INFO exited: web (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-08-09 02:06:54,105 INFO spawned: 'web' with pid 212
hadabot_ros2_codeserver  | Traceback (most recent call last):
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 122, in <module>
hadabot_ros2_codeserver  |     main()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 118, in main
hadabot_ros2_codeserver  |     entrypoint()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 78, in run_server
hadabot_ros2_codeserver  |     from vnc.app import app
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/vnc/app.py", line 6, in <module>
hadabot_ros2_codeserver  |     from flask import (
hadabot_ros2_codeserver  |   File "/usr/local/lib/python3.8/dist-packages/flask/__init__.py", line 14, in <module>
hadabot_ros2_codeserver  |     from jinja2 import escape
hadabot_ros2_codeserver  | ImportError: cannot import name 'escape' from 'jinja2' (/home/hadabot/.local/lib/python3.8/site-packages/jinja2/__init__.py)
hadabot_ros2_codeserver  | 2023-08-09 02:06:54,757 INFO exited: web (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-08-09 02:06:55,763 INFO gave up: web entered FATAL state, too many start retries too quickly`

coltonwestern, 2023-12-28:
I am having the same issue here as well. Has anyone found a solution to this error? 

Attaching to hadabot_ros2_codeserver, hadabot_ros2_web_bridge
hadabot_ros2_codeserver  | * enable custom user: hadabot
hadabot_ros2_codeserver  | useradd: user 'hadabot' already exists
hadabot_ros2_codeserver  |   set default password to "ubuntu"
hadabot_ros2_codeserver  | cp: cannot stat '/root/.config': No such file or directory
hadabot_ros2_web_bridge  | Websocket started on ws://localhost:9090
hadabot_ros2_codeserver  | 2023-12-28 07:14:48,736 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
hadabot_ros2_codeserver  | 2023-12-28 07:14:48,736 INFO Included extra file "/etc/supervisor/conf.d/codeserver.conf" during parsing
hadabot_ros2_codeserver  | 2023-12-28 07:14:48,736 INFO Included extra file "/etc/supervisor/conf.d/supervisord.conf" during parsing
hadabot_ros2_codeserver  | 2023-12-28 07:14:48,744 INFO RPC interface 'supervisor' initialized
hadabot_ros2_codeserver  | 2023-12-28 07:14:48,744 CRIT Server 'unix_http_server' running without any HTTP authentication checking
hadabot_ros2_codeserver  | 2023-12-28 07:14:48,745 INFO supervisord started with pid 16
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,749 INFO spawned: 'nginx' with pid 18
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,755 INFO spawned: 'web' with pid 19
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,761 INFO spawned: 'codeserver' with pid 20
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,767 INFO spawned: 'xvfb' with pid 21
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,774 INFO spawned: 'wm' with pid 22
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,781 INFO spawned: 'lxpanel' with pid 26
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,785 INFO spawned: 'pcmanfm' with pid 29
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,788 INFO spawned: 'x11vnc' with pid 32
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,793 INFO spawned: 'novnc' with pid 34
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,794 INFO exited: xvfb (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,806 INFO exited: pcmanfm (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,807 INFO exited: lxpanel (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,810 INFO exited: wm (exit status 1; not expected)
hadabot_ros2_codeserver  | Traceback (most recent call last):
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 122, in <module>
hadabot_ros2_codeserver  |     main()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 118, in main
hadabot_ros2_codeserver  |     entrypoint()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 78, in run_server
hadabot_ros2_codeserver  |     from vnc.app import app
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/vnc/app.py", line 6, in <module>
hadabot_ros2_codeserver  |     from flask import (
hadabot_ros2_codeserver  |   File "/usr/local/lib/python3.8/dist-packages/flask/__init__.py", line 14, in <module>
hadabot_ros2_codeserver  |     from jinja2 import escape
hadabot_ros2_codeserver  | ImportError: cannot import name 'escape' from 'jinja2' (/home/hadabot/.local/lib/python3.8/site-packages/jinja2/__init__.py)
hadabot_ros2_codeserver  | 2023-12-28 07:14:49,992 INFO exited: web (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,803 INFO success: nginx entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,804 INFO success: codeserver entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,807 INFO spawned: 'xvfb' with pid 77
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,813 INFO spawned: 'lxpanel' with pid 78
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,820 INFO spawned: 'pcmanfm' with pid 80
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,822 INFO success: x11vnc entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,822 INFO success: novnc entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,823 INFO exited: x11vnc (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,829 INFO spawned: 'wm' with pid 82
hadabot_ros2_codeserver  | 2023-12-28 07:14:50,837 INFO spawned: 'x11vnc' with pid 84
hadabot_ros2_codeserver  | 2023-12-28 07:14:51,001 INFO spawned: 'web' with pid 123
hadabot_ros2_codeserver  | Traceback (most recent call last):
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 122, in <module>
hadabot_ros2_codeserver  |     main()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 118, in main
hadabot_ros2_codeserver  |     entrypoint()
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/run.py", line 78, in run_server
hadabot_ros2_codeserver  |     from vnc.app import app
hadabot_ros2_codeserver  |   File "/usr/local/lib/web/backend/vnc/app.py", line 6, in <module>
hadabot_ros2_codeserver  |     from flask import (
hadabot_ros2_codeserver  |   File "/usr/local/lib/python3.8/dist-packages/flask/__init__.py", line 14, in <module>
[... cut ...]
hadabot_ros2_codeserver  |   File "/usr/local/lib/python3.8/dist-packages/flask/__init__.py", line 14, in <module>
hadabot_ros2_codeserver  |     from jinja2 import escape
hadabot_ros2_codeserver  | ImportError: cannot import name 'escape' from 'jinja2' (/home/hadabot/.local/lib/python3.8/site-packages/jinja2/__init__.py)
hadabot_ros2_codeserver  | 2023-12-28 07:14:57,306 INFO exited: web (exit status 1; not expected)
hadabot_ros2_codeserver  | 2023-12-28 07:14:58,308 INFO gave up: web entered FATAL state, too many start retries too quickly

bsplonka, 2024-04-27:
I had the same problems, I got further, by using
`pip install Flask==2.2.2`
and
`pip install Werkzeug==2.3.7`
[https://stackoverflow.com/questions/77213053/why-did-flask-start-failing-with-importerror-cannot-import-name-url-quote-fr]
However, I then get the following error:
subprocess.calledprocesserror: command '['supervisorctl', '-c', '/etc/supervisor/supervisord.conf', 'status']' returned non-zero exit status 3.