hadabot/hadabot_main #35: Building Dockerfile.ros2_codeserver fails with `permission denied` on 27/35 and 29/35

mvstinger, 2023-08-12:
...in both/either `docker-compose up` or `docker build`; (e.g.) On Pop!_OS 22.04 LTS:

## Step 27/35 (cpptools, python extensions)
```bash
Step 27/35 : RUN wget https://github.com/microsoft/vscode-cpptools/releases/download/1.0.1/cpptools-linux.vsix &&     wget https://github.com/microsoft/vscode-python/releases/download/2020.5.86806/ms-python-release.vsix &&     code-server --verbose --install-extension cpptools-linux.vsix &&     code-server --verbose --install-extension ms-python-release.vsix &&     rm cpptools-linux.vsix &&     rm ms-python-release.vsix &&     code-server --install-extension twxs.cmake &&     /usr/bin/python3 -m pip install -U numpy mypy autopep8 flake8 jupyterlab pandas transforms3d scipy --user &&     echo "source /usr/share/virtualenvwrapper/virtualenvwrapper.sh" >> .bashrc
 ---> Running in c26d1be32a84
--2023-08-12 12:28:49--  https://github.com/microsoft/vscode-cpptools/releases/download/1.0.1/cpptools-linux.vsix
Resolving github.com (github.com)... 140.82.112.3
Connecting to github.com (github.com)|140.82.112.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/54800346/619ead80-fc25-11ea-8c66-848e23495edb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230812%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230812T122849Z&X-Amz-Expires=300&X-Amz-Signature=bea3db3b5c7a794e7153df3bb2e90a9b49e84190319707d14ba172b6d0357951&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=54800346&response-content-disposition=attachment%3B%20filename%3Dcpptools-linux.vsix&response-content-type=application%2Foctet-stream [following]
--2023-08-12 12:28:49--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/54800346/619ead80-fc25-11ea-8c66-848e23495edb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230812%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230812T122849Z&X-Amz-Expires=300&X-Amz-Signature=bea3db3b5c7a794e7153df3bb2e90a9b49e84190319707d14ba172b6d0357951&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=54800346&response-content-disposition=attachment%3B%20filename%3Dcpptools-linux.vsix&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 21192772 (20M) [application/octet-stream]
cpptools-linux.vsix: Permission denied

Cannot write to ‘cpptools-linux.vsix’ (Success).
The command '/bin/sh -c wget https://github.com/microsoft/vscode-cpptools/releases/download/1.0.1/cpptools-linux.vsix &&     wget https://github.com/microsoft/vscode-python/releases/download/2020.5.86806/ms-python-release.vsix &&     code-server --verbose --install-extension cpptools-linux.vsix &&     code-server --verbose --install-extension ms-python-release.vsix &&     rm cpptools-linux.vsix &&     rm ms-python-release.vsix &&     code-server --install-extension twxs.cmake &&     /usr/bin/python3 -m pip install -U numpy mypy autopep8 flake8 jupyterlab pandas transforms3d scipy --user &&     echo "source /usr/share/virtualenvwrapper/virtualenvwrapper.sh" >> .bashrc' returned a non-zero code: 3
```

## Step 29/35 (ros extension):
```bash
Step 29/35 : RUN unzip vscode-ros-dev-vsix.zip &&     code-server --install-extension vscode-ros-dev.vsix &&     rm vscode-ros-dev*.*
 ---> Running in 188630e583a7
Archive:  vscode-ros-dev-vsix.zip
error:  cannot create vscode-ros-dev.vsix
        Permission denied
The command '/bin/sh -c unzip vscode-ros-dev-vsix.zip &&     code-server --install-extension vscode-ros-dev.vsix &&     rm vscode-ros-dev*.*' returned a non-zero code: 50
```

mvstinger, 2023-08-12:
No rights to branch and make pull request- the following allows build to complete and run the vscode web UI (YMMV: Not tested further!!):

In docker/Dockerfile.ros2_codeserver move lines 126:130 to 147:

```
USER hadabot

WORKDIR /home/hadabot
ENV HOME=$WORKDIR
```

jackpien, 2023-08-16:
Hi @mvstinger - 

> No rights to branch and make pull request

That's strange - the repo should not be protected. Are you not able to click on "Pull Requests" above and create a new pull request?

Thanks
Jack